chilipeppr/widget-eagle #5: Dimension layer object not getting milled.

momiccioli, 2016-02-17:
Chilipeppr is creating gcode for the outer most profile on the dimension layer.  As a result drills on and other items on the dimension layer will not get encoded in the gcode. 

The work around for this is to create vias in place of holes.

chilipeppr, 2016-02-17:
What does the xml look like in the board file for the holes?

momiccioli, 2016-02-17:
> <reply+01080a2dcb07e5d7cfc6a734f3ed2138c0e54e52e9a57f5192cf0000000112dcc2719
> 2a169ce08034090@reply.github.com>

Here is the xml of one of the holes.

<packages>
<package name="STAND-OFF">
<description>&lt;b&gt;Stand Off&lt;/b&gt;&lt;p&gt;
This is the mechanical footprint for a #4 phillips button head screw. Use
the keepout ring to avoid running the screw head into surrounding
components. SKU : PRT-00447</description>
<wire x1="0" y1="1.8542" x2="0" y2="-1.8542" width="0.2032" layer="41"
curve="-180"/>
<wire x1="0" y1="-1.8542" x2="0" y2="1.8542" width="0.2032" layer="41"
curve="-180"/>
<wire x1="0" y1="-1.8542" x2="0" y2="1.8542" width="0.2032" layer="42"
curve="180"/>
<wire x1="0" y1="-1.8542" x2="0" y2="1.8542" width="0.2032" layer="42"
curve="-180"/>
<circle x="0" y="0" radius="2.794" width="0.127" layer="39"/>
<hole x="0" y="0" drill="3.302"/>
</package>
</packages>

chilipeppr, 2016-02-17:
That's going to take some work to get that in there. You'd have to parse
the dimensions and realize there's a reference to a package, then analyze
the package objects to render them including appending the drill array
which will need milling or drilling options. Up for helping out?

raykholo, 2016-02-18:
Guys, 
I went through the eagle widget workflow in console with (a close enough one to) this board.  

The XML that @momiccioli pasted above is the top of the eagle file where the relevant components from all the libraries are restated in an effort to make the eagle files mobile without external library dependencies.  

The good news is that the parser does indeed pick up on this standoff as it is an element:
`Object {pkg: "STAND-OFF", name: "U$1", value: "", x: 3.175, y: 17.93875…}attributes: Objectmatrix: Array[4]mirror: falsename: "U$1"padSignals: Object__proto__: Objectpkg: "STAND-OFF"rot: "R0"smashed: nullthreeObj: Objectpadgroup: THREE.Object3Dpads: ObjectpadsAsLines: ObjectpadsAsMesh: Objectsmdgroup: THREE.Object3Dsmds: ObjectsmdsClipperFmt: Object__proto__: Objectvalue: ""x: 3.175y: 17.93875__proto__: Object`

We're not interested in the wires.  Layers 39, 41, and 42 are restrict and keepout layers for the autorouter.  We are interested in the hole attribute right at the end.  It's just another case to add into draw3dElements.  I have not yet determined if Eagle Parse passes this information though.

I will have to check how this works with a hole you make yourself, as opposed to this example which was an actual component from the Sparkfun library.  

As for the holes from the RJ45 jack mounting, same drill  (no pun intended). Please see the bottom of this packages. 
`<packages>
<package name="RJ45-8">
<wire x1="-7.62" y1="10" x2="7.62" y2="10" width="0.2032" layer="21"/>
<wire x1="-7.62" y1="-3" x2="-7.62" y2="10" width="0.2032" layer="21"/>
<wire x1="-7.62" y1="-3" x2="7.62" y2="-3" width="0.2032" layer="21"/>
<wire x1="7.62" y1="10" x2="7.62" y2="-3" width="0.2032" layer="21"/>
<wire x1="-7.62" y1="-3" x2="-7.62" y2="-5.72" width="0.2032" layer="51"/>
<wire x1="-7.62" y1="-5.72" x2="7.62" y2="-5.72" width="0.2032" layer="51"/>
<wire x1="7.62" y1="-5.72" x2="7.62" y2="-3" width="0.2032" layer="51"/>
<pad name="8" x="4.445" y="8.89" drill="1" diameter="1.8796"/>
<pad name="7" x="3.175" y="6.35" drill="1" diameter="1.8796"/>
<pad name="6" x="1.905" y="8.89" drill="1" diameter="1.8796"/>
<pad name="4" x="-0.635" y="8.89" drill="1" diameter="1.8796"/>
<pad name="2" x="-3.175" y="8.89" drill="1" diameter="1.8796"/>
<pad name="5" x="0.635" y="6.35" drill="1" diameter="1.8796"/>
<pad name="3" x="-1.905" y="6.35" drill="1" diameter="1.8796"/>
<pad name="1" x="-4.445" y="6.35" drill="1" diameter="1.8796" shape="square"/>
<hole x="-5.715" y="0" drill="3.2"/>
<hole x="5.715" y="0" drill="3.2"/>
</package>`

@momiccioli please upload the .brd file somewhere and provide the link here, as I do not see a way to attach a file directly.

momiccioli, 2016-02-18:
Here is the board file that I just shared with Ray.  It has the holes from the RJ45 and standoffs.  I created a hole at location 50, 725 with diameter 25 (mils).  Attached is the board file.

[Dual RJ45 Board 1.brd.zip](https://github.com/chilipeppr/widget-eagle/files/135476/Dual.RJ45.Board.1.brd.zip)

ameennihad, 2017-01-03:
This issue has be solved, to be released with Version 5.2

ameennihad, 2017-01-03:
If anyone want to test, please check the fixed code in this workspace http://www.chilipeppr.com/ameen

momiccioli, 2017-01-03:
> <reply+01080a2dd678ac11b63308eb2723dd4176bc1eb81f46ce3092cf000000011483c5e49

I am running a test of the Mirroring right now and all looks good.  I just
did an export of the Gcode into my Gcode editor and the hole looks perfect.
Once I finish the routing I¹ll confirm back to you.

Date:  Tuesday, January 3, 2017 at 3:04 PM
Cc:  Michael Omiccioli <michael@omiccioli.com>, Mention
<mention@noreply.github.com>
milled. (#5)

http://www.chilipeppr.com/ameen


WRN2dirks5rOqnkgaJpZM4Hcm2E> .

  
b","name":"GitHub"},"entity":{"external_key":"github/chilipeppr/widget-eagle
","title":"chilipeppr/widget-eagle","subtitle":"GitHub
repository","main_image_url":"https://cloud.githubusercontent.com/assets/143
418/17495839/a5054eac-5d88-11e6-95fc-7290892c7bb5.png","avatar_image_url":"h
ttps://cloud.githubusercontent.com/assets/143418/15842166/7c72db34-2c0b-11e6
-9aed-b52498112777.png","action":{"name":"Open in
GitHub","url":"https://github.com/chilipeppr/widget-eagle"}},"updates":{"sni
ppets":[{"icon":"PERSON","message":"@ameennihad in #5: If anyone want to
http://www.chilipeppr.com/ameen"}],"action":{"name":"View
Issue","url":"https://github.com/chilipeppr/widget-eagle/issues/5#issuecomme
nt-270210316"}}}

momiccioli, 2017-01-03:
One thing I did notice (not sure if this is new or not) is the milling path for the cutout of my board does not have rounded corners.  They are diagonal even though I have rounded corners in my Eagle design.  This is not a huge issue but I though I would report it.  I can add a new feature request if desired.

chilipeppr, 2017-01-03:
Yeah, add a feature request, and please put in a screenshot and attach
sample files if you could. My guess is that Eagle has an extra parameter we
haven't seen/looked for and we'd have to render a circle into multiple mini
lines. Auto-level can only level G1 moves which are straight lines, so
although a G2 curve would be nice, those are hard to deal with.

momiccioli, 2017-01-03:
> <reply+01080a2db60499dd111b1a6ffe88e1dca25842e47bb670d492cf000000011483c9ab9

John,

Your spot on with the diagnosis, I wrote a Python script to create that code
and it was a pain to get right.  Its a G2 or G3 depending on the direction
of travel.  Ill share that code too.  If I had more time to come out to
speed on the development environment I would do it myself.  Just too much
going on.  Hope I can contribute this way though.  Ill test though.

Date:  Tuesday, January 3, 2017 at 3:20 PM
milled. (#5)

7ZGks5rOq0dgaJpZM4Hcm2E>


hhykHAbks5rOq2rgaJpZM4Hcm2E> .

  
ppets":[{"icon":"PERSON","message":"@chilipeppr in #5: Yeah, add a feature
request, and please put in a screenshot and attach\nsample files if you
could. My guess is that Eagle has an extra parameter we\nhaven't seen/looked
for and we'd have to render a circle into multiple mini\nlines. Auto-level
can only level G1 moves which are straight lines, so\nalthough a G2 curve
would be nice, those are hard to deal with.\n\nOn Tue, Jan 3, 2017 at 12:18
PM, Michael Omiccioli \u003cnotifications@github.com\n\u003e
wrote:\n\n\u003e One thing I did notice (not sure if this is new or not) is
the milling\n\u003e path for the cutout of my board does not have rounded
corners. They are\n\u003e diagonal even though I have rounded corners in my
Eagle design. This is not\n\u003e a huge issue but I though I would report
it. I can add a new feature\n\u003e request if desired.\n\u003e\n\u003e
this email directly, view it on GitHub\n\u003e
\u003chttps://github.com/chilipeppr/widget-eagle/issues/5#issuecomment-27021
jmVsN4Of47ZGks5rOq0dgaJpZM4Hcm2E\u003e\n\u003e
.\n\u003e\n"}],"action":{"name":"View
Issue","url":"https://github.com/chilipeppr/widget-eagle/issues/5#issuecomme
nt-270213996"}}}

chilipeppr, 2017-01-03:
> 270213420>,
> AHidbQ1JqP9eZReYJPBcjmVsN4Of4
> auth/AQgKLScl7MNq1W_x7TeFF0p6S
> ,"updates":{"sni
> issues/5#issuecomment-27021
> auth/AHidbQ1JqP9eZReYJPBc
> issues/5#issuecomme

You should watch the 3 videos on the ChiliPeppr homepage to see how clean
and easy the ChiliPeppr structure is. It might give you some good ideas.
We'd love your help on the Eagle widget if you were able to. It's getting
really good.

momiccioli, 2017-01-03:
> <reply+01080a2dfafb03411cd33d30038060efb93d66459ef865a392cf000000011483d0d09

I plan to take another run at it.  I might a few interactive sessions to
ensure I understand the various components.  There are a bunch of things I
would like to see and would love to contribute.

Date:  Tuesday, January 3, 2017 at 3:51 PM
milled. (#5)

c8yks5rOq5pgaJpZM4Hcm2E>

72tFk2bks5rOrTQgaJpZM4Hcm2E> .

  
ppets":[{"icon":"PERSON","message":"@chilipeppr in #5: You should watch the
3 videos on the ChiliPeppr homepage to see how clean\nand easy the
ChiliPeppr structure is. It might give you some good ideas.\nWe'd love your
help on the Eagle widget if you were able to. It's getting\nreally
good.\n\nOn Tue, Jan 3, 2017 at 12:24 PM, Michael Omiccioli

\n\n\u003e
John,\n\u003e\n\u003e Your spot on with the diagnosis, I wrote a Python
script to create that\n\u003e code\n\u003e and it was a pain to get right.
Its a G2 or G3 depending on the direction\n\u003e of travel. Ill share that
code too. If I had more time to come out to\n\u003e speed on the development
environment I would do it myself. Just too much\n\u003e going on. Hope I can
contribute this way though. Ill test though.\n\u003e\n\u003e From: John
chilipeppr/widget-eagle\n\u003e
\u003creply+01080a2db60499dd111b1a6ffe88e1dca25842e47bb670d492cf00000001\n\u
003e 1483c9ab9\n\u003e 2a169ce08034090@reply.github.com\u003e\n\u003e Date:
Tuesday, January 3, 2017 at 3:20 PM\n\u003e To: chilipeppr/widget-eagle
\u003cwidget-eagle@noreply.github.com\u003e\n\u003e Cc: Michael Omiccioli
\u003cmichael@omiccioli.com\u003e, Mention\n\u003e
[chilipeppr/widget-eagle] Dimension layer object not getting\n\u003e milled.
(#5)\n\u003e\n\u003e Yeah, add a feature request, and please put in a
screenshot and attach\n\u003e sample files if you could. My guess is that
Eagle has an extra parameter we\n\u003e haven't seen/looked for and we'd
have to render a circle into multiple mini\n\u003e lines. Auto-level can
only level G1 moves which are straight lines, so\n\u003e although a G2 curve
would be nice, those are hard to deal with.\n\u003e\n\u003e On Tue, Jan 3,
2017 at 12:18 PM, Michael Omiccioli \u003c\n\u003e

\n\u003e\n\u003e \u003e One
are\n\u003e \u003e diagonal even though I have rounded corners in my Eagle
design. This is\n\u003e not\n\u003e \u003e a huge issue but I though I would
report it. I can add a new feature\n\u003e \u003e request if
desired.\n\u003e \u003e\n\u003e \u003e ‹\n\u003e \u003e You are receiving
view it on GitHub\n\u003e \u003e
\u003chttps://github.com/chilipeppr/widget-eagle/issues/5#issuecomment-\n\u0
\u003e\n\u003e 
AHidbQ1JqP9eZReYJPBcjmVsN4Of4\n\u003e 7ZGks5rOq0dgaJpZM4Hcm2E\u003e\n\u003e
\u003e .\n\u003e \u003e\n\u003e\n\u003e ‹\n\u003e You are receiving this
auth/AQgKLScl7MNq1W_x7TeFF0p6S\n\u003e hhykHAbks5rOq2rgaJpZM4Hcm2E\u003e
.\n\u003e\n\u003e\n\u003e
{\"api_version\":\"1.0\",\"publisher\":{\"api_key\":\"\n\u003e
05dde50f1d1a384dd78767c55493e4b\n\u003e
b\",\"name\":\"GitHub\"},\"entity\":{\"external_key\":\"github/\n\u003e
chilipeppr/widget-eagle\n\u003e
\",\"title\":\"chilipeppr/widget-eagle\",\"subtitle\":\"GitHub\n\u003e
/\n\u003e 143\n\u003e
418/17495839/a5054eac-5d88-11e6-95fc-7290892c7bb5.png\",\"\n\u003e
avatar_image_url\":\"h\n\u003e
ttps://cloud.githubusercontent.com/assets/143418/15842166/7c72db34-2c0b-\n\u
003e 11e6\n\u003e -9aed-b52498112777.png\",\"action\":{\"name\":\"Open
in\n\u003e 
GitHub\",\"url\":\"https://github.com/chilipeppr/widget-eagle\"}}\n\u003e
,\"updates\":{\"sni\n\u003e
feature\n\u003e request, and please put in a screenshot and attach\\nsample
files if you\n\u003e could. My guess is that Eagle has an extra parameter
we\\nhaven't\n\u003e seen/looked\n\u003e for and we'd have to render a
circle into multiple mini\\nlines. Auto-level\n\u003e can only level G1
moves which are straight lines, so\\nalthough a G2 curve\n\u003e would be
nice, those are hard to deal with.\\n\\nOn Tue, Jan 3, 2017 at 12:18\n\u003e
PM, Michael Omiccioli \\u003cnotifications@github.com\\n\\u003e\n\u003e
is\n\u003e the milling\\n\\u003e path for the cutout of my board does not
have rounded\n\u003e corners. They are\\n\\u003e diagonal even though I have
rounded corners in my\n\u003e Eagle design. This is not\\n\\u003e a huge
issue but I though I would report\n\u003e it. I can add a new
to\n\u003e this email directly, view it on GitHub\\n\\u003e\n\u003e
\\u003chttps://github.com/chilipeppr/widget-eagle/\n\u003e
issues/5#issuecomment-27021\n\u003e 3420\\u003e,\\n\\u003e or mute the
thread\\n\\u003e\n\u003e
auth/AHidbQ1JqP9eZReYJPBc\n\u003e
.\\n\\u003e\\n\"}],\"action\":{\"name\":\"View\n\u003e
Issue\",\"url\":\"https://github.com/chilipeppr/widget-eagle/\n\u003e
issues/5#issuecomme\n\u003e nt-270213996\"}}}\n\u003e\n\u003e —\n\u003e You
are receiving this because you were mentioned.\n\u003e Reply to this email
nv9PW-NS4c8yks5rOq5pgaJpZM4Hcm2E\u003e\n\u003e
nt-270221193"}}}

closed 2017-01-14 by ameennihad