chilipeppr/widget-eagle #6: Bottom layer import from Eagle needs to be mirrored

momiccioli, 2016-02-17:
When circuit designs are imported to CP from Eagle the bottom layer will be encoded as if it were the top layer.  Chilipeppr will need to create a mirror of the bottom layer before rendering and encoding to gcode.

A work around is to create a mirror of the bottom layer in eagle and then import it into Chilipeppr.  Chilipeppr will then render the board correctly as if it were the top layer.  There are videos that explain this on the Chilipeppr hangout.

ameennihad, 2017-01-03:
This feature is now implemented in version 5.1 of the widget, I believe we should mark this issue as solved.

closed 2017-01-03 by chilipeppr

momiccioli, 2017-01-03:
Excellent news I will test this in the next hour and confirm.

momiccioli, 2017-01-03:
Confirmed this works!!!!!  Yahoo.