chilipeppr/widget-eagle #7: Include tabs in permiter mill of circuit boards

momiccioli, 2016-02-17:
This is a feature request,  The ability to have tabs created as part of the perimeter mill of the board. These tabs will ensure the board stays in place during the final layers of the mill.  Otherwise the board could come loose and be damaged if it gets tangled up in the spindle.

ameennihad, 2017-01-03:
Frank just pulled this feature from xpix workspace widget.

closed 2017-01-03 by ameennihad

chilipeppr, 2017-01-03:
The tabs look weird on the Arduino 2-layer board. And, the top cut on the
lower right seems to be gone. Hmm.
[image: Inline image 1]

Playing with the settings you can get something cleaner, but that last
segment seems lost of the top mill path in lower right corner of board in
screenshot.

[image: Inline image 2]

ameennihad, 2017-01-04:
It seems that the screenshots were not uploaded!
I'm guessing that this is related to board dimensions bug, let's test after bug fix is pulled to main project.