chilipeppr/widget-eagle #11: Dimnsions rendered incorrectly

ameennihad, 2017-01-03:
Eagle lists dimension wires in the order the user draws them, however widget code assumes end of each wire matches the start of next one, and end of last wire matches the start of first one, which is true of most of the board but not all.
Following screenshot is taken from xpix workspace just to show that it's an old bug and not related to recent updates, it renders exactly the same with Version 5.1 of the widget (of course without the nice tabs that Frank Herrmann made)
[image]

chilipeppr, 2017-01-03:
What is generating those tabs for the dimensions? That's an awesome
feature. Was that something Frank added? Or did you add that?

ameennihad, 2017-01-03:
Frank added that not me, really nice feature, can we pull from his repository? if not I can add his code with my next update.

chilipeppr, 2017-01-03:
Hi Ameen, I just made you a collaborator on the Github project, so you
should be able to do more stuff as you see fit. Thanks for all of your
amazing help here. It's really great stuff. I'm already rethinking my
boards now that we can do double-sided. I'm excited about all the new
features and indeed these tabs would be awesome to have in the mix too.
This widget is getting really good.

xpix, 2017-01-03:
Hello Friends, yes this change is needed for XDisPlace, first i cut the PCB and need this holder for further drill and mill. ok, i check my repository to find this changes and make a pull request.

xpix, 2017-01-03:
PLease check this out:
https://github.com/chilipeppr/widget-eagle/pull/12

chilipeppr, 2017-01-03:
BTW, both Frank and Ameen are now collaborators on Github for widget-eagle,
so feel free to do pull requests to the main branch, but I would like to
always check through them as well to catch anything that might look like it
could break something.

chilipeppr, 2017-01-03:
Scanned through the code. Looks great and nothing worries me. Frank, do you
want to try to complete the pull request yourself then now that you are a
Collaborator?

ameennihad, 2017-01-03:
I fixed board dimensions bug, please check my experimental workspace http://www.chilipeppr.com/ameen
Please try any weird shaped boards you might have.

chilipeppr, 2017-01-03:
That's awesome. Even the Adafruit Ruler PCB has a dimension now. Prior to
your bug fix nothing was showing up for that. Wow, nice work again.
Incredible progress.

ameennihad, 2017-01-03:
Code still needs some improvements, I just created a test board with complex dimensions, I see one wire went in the wrong direction, I'll check that tomorrow.

chilipeppr, 2017-01-03:
How did you do it? Match up the x,y on each wire?

ameennihad, 2017-01-03:
Yes, exactly.

ameennihad, 2017-01-04:
I think I solved this bug for good, see before and after screenshots of the weirdest board I ever designed :), next steps is to handle cures in board dimensions.

**Before:**
[image]

**After:**
[image]

xpix, 2017-01-04:
Great News and welcome in our Team! :)

chilipeppr, 2017-01-04:
Nice. That looks like you nailed it.

closed 2017-01-14 by ameennihad