chilipeppr/widget-eagle #13: Cut out milling not following corner radius for rounded corners.

momiccioli, 2017-01-03:
When the cut dimensions of the PCB board are rounded the Gcode produced does not follow the radius of the corner.  In the Plane section of the .brd file when there is a rounded corner the wire line includes a curve parameter -90 degrees for clockwise rotation and 90 degrees for counter clockwise.   If you do not process the curve parameter and travel to the start point of the next wire you get an angled corner.  

 Picture of the brd in Eagle
[image]

Picture of widget Gcode results, see how the corners are clipped not rounded.
[image]

Picture of how the Gcode should look with rounded corners.  The code below produced this.
[image]

The answer to this problem is to use G02 and G03 in relation to the curve value and the start stop X and Y coordinates.  I must admit I am not an expert at Gcode or Eagle XML but the code below has been working for me.  It's python code but very straight forward.

Assumptions:
  CurveIt is called using the variables for the current wire if there is a curve item.

  The I and J values are determined by the value of the curve (-90 or 90). The relationship of
   X1/Y1 to X2/Y2.  There may be a better way to do this ( I just hate geometry).
  
```
#-------------------------------------------------------------------------------
def CurveIt(x1,y1,x2,y2, crv):
    ij = abs(x1 - x2)
    if x1 > x2 and y1 < y2 :
        if crv == '-90' :
            res = "G02 X" + str(x2) + " Y" + str(y2) + " I0 J" + str(ij)
        else:
            res = "G03 X" + str(x2) + " Y" + str(y2) + " I-"  + str(ij)+ " J0"

    if x1 < x2 and y1 < y2 :
        if crv == '-90' :
            res = "G02 X" + str(x2) + " Y" + str(y2) + " I" + str(ij) + " J0"
        else:
            res = "G03 X" + str(x2) + " Y" + str(y2) + " I0 J" + str(ij)

    if x1 < x2 and y1 > y2 :
        if crv == '-90' :
            res = "G02 X" + str(x2) + " Y" + str(y2) + " I0 J-" + str(ij)
        else:
            res = "G03 X" + str(x2) + " Y" + str(y2) + " I" + str(ij) + " J0"

    if x1 > x2 and y1 > y2 :
        if crv == '-90' :
            res = "G02 X" + str(x2) + " Y" + str(y2) + " I-"  + str(ij)+ " J0"
        else:
            res = "G03 X" + str(x2) + " Y" + str(y2) + " I0 J-" + str(ij)

    return res

#-------------------------------------------------------------------------------
```

ameennihad, 2017-01-04:
This a known limitation, currently curve parameters are ignored, its in my to do list, I expect to fix it shortly.

ameennihad, 2017-01-05:
Board dimensions with curves are now supported, still experimental I need to do further testing, I implemented the curves as line segments (currently 8 per each 1/4 circle), so no G02/G03 is generated, I might make the number of segments dynamic depending on curve size.
I decided to take this approach for following reasons:
- I didn't want to mess-up with clipperDimension array, it's used in different parts of the widget, if I want to use G02/G03 I have to add a third parameter to each vertex in the array.
- In my approach I didn't touch the Gcode generation functions, if I want to use G02/G03 I have to rewrite that and possibly other inflation functions.

Here are two screenshots, you can test the feature on my test workspace, link in screenshots:
[image]

[image]

momiccioli, 2017-01-05:
> <reply+01080a2d3c40e56d68bb18d4041e4b30eb8c0e59abc4716192cf000000011485e1a29
> 2a169ce0bd61083@reply.github.com>
> <author@noreply.github.com>
> radius for rounded corners. (#13)

I ran a test of a board design into my Gcode editor and it looks good.  If
that is happy then the mill is good.  I will try to do a real test later
today but it looks good.

 <https://cloud.githubusercontent.com/assets/24837471/21677191/18825c1c-d34a
-11e6-96d6-40b96f952ad5.png>

 <https://cloud.githubusercontent.com/assets/24837471/21677197/1f6a0eb2-d34a
-11e6-9f69-f306fcf90079.png>


<https://github.com/chilipeppr/widget-eagle/issues/13#issuecomment-270612423
Bkt9F3pks5rPMWigaJpZM4LaEPs> .

  
b","name":"GitHub"},"entity":{"external_key":"github/chilipeppr/widget-eagle
","title":"chilipeppr/widget-eagle","subtitle":"GitHub
repository","main_image_url":"https://cloud.githubusercontent.com/assets/143
418/17495839/a5054eac-5d88-11e6-95fc-7290892c7bb5.png","avatar_image_url":"h
ttps://cloud.githubusercontent.com/assets/143418/15842166/7c72db34-2c0b-11e6
-9aed-b52498112777.png","action":{"name":"Open in
GitHub","url":"https://github.com/chilipeppr/widget-eagle"}},"updates":{"sni
ppets":[{"icon":"PERSON","message":"@ameennihad in #13: Board dimensions
dynamic depending on curve size.\r\nI decided to take this approach for
following reasons:\r\n- I didn't want to mess-up with clipperDimension
I have to add a third parameter to each vertex in the array.\r\n- In my
functions.\r\n\r\nHere are two screenshots, you can test the feature on my
test workspace, link in
screenshots:\r\n![curves_board](https://cloud.githubusercontent.com/assets/2
4837471/21677191/18825c1c-d34a-11e6-96d6-40b96f952ad5.png)\r\n\r\n![curves_g
code](https://cloud.githubusercontent.com/assets/24837471/21677197/1f6a0eb2-
d34a-11e6-9f69-f306fcf90079.png)\r\n"}],"action":{"name":"View
Issue","url":"https://github.com/chilipeppr/widget-eagle/issues/13#issuecomm
ent-270612423"}}}

chilipeppr, 2017-01-05:
> auth/AQgKLXJo319jUJl8IeYGWIlhs
> ,"updates":{"sni
> githubusercontent.com/assets/2
> 24837471/21677197/1f6a0eb2-
> issues/13#issuecomm

Yeah, agreed on not using G2/G3. You can't inflate/deflate those paths
using Clipper and thus all sorts of stuff falls apart. I agree with your
approach. The curvy corners look great. Once again, amazing work Ameen.
You're incredible.
-John

momiccioli, 2017-01-05:
> <reply+01080a2dd2bded6cf47b099d5cf327dfbbcbca0819ad497592cf000000011486434b9

Agreed excellent work and I like the results.

Date:  Thursday, January 5, 2017 at 12:24 PM

FjTks5rPQjWgaJpZM4LaEPs>

<https://github.com/chilipeppr/widget-eagle/issues/13#issuecomment-270703008
CHnKfo-ks5rPSdLgaJpZM4LaEPs> .

  
ppets":[{"icon":"PERSON","message":"@chilipeppr in #13: Yeah, agreed on not
using G2/G3. You can't inflate/deflate those paths\nusing Clipper and thus
all sorts of stuff falls apart. I agree with your\napproach. The curvy
corners look great. Once again, amazing work Ameen.\nYou're
incredible.\n-John\n\nOn Thu, Jan 5, 2017 at 7:14 AM, Michael Omiccioli

\n\n\u003e I ran a test of a
board design into my Gcode editor and it looks good. If\n\u003e that is
happy then the mill is good. I will try to do a real test later\n\u003e
today but it looks good.\n\u003e\n\u003e From: ameennihad
chilipeppr/widget-eagle\n\u003e
\u003creply+01080a2d3c40e56d68bb18d4041e4b30eb8c0e59abc4716192cf00000001\n\u
003e 1485e1a29\n\u003e 2a169ce0bd61083@reply.github.com\u003e\n\u003e Date:
Thursday, January 5, 2017 at 5:27 AM\n\u003e To: chilipeppr/widget-eagle
\u003cwidget-eagle@noreply.github.com\u003e\n\u003e Cc: Michael Omiccioli
\u003cmichael@omiccioli.com\u003e, Author\n\u003e
[chilipeppr/widget-eagle] Cut out milling not following corner\n\u003e
radius for rounded corners. (#13)\n\u003e\n\u003e\n\u003e Board dimensions
with curves are now supported, still experimental I need\n\u003e to\n\u003e
8\n\u003e per each 1/4 circle), so no G02/G03 is generated, I might make the
number\n\u003e of\n\u003e segments dynamic depending on curve size.\n\u003e
I decided to take this approach for following reasons:\n\u003e * I didn't
want to mess-up with clipperDimension array, it's used in\n\u003e different
parts of the widget, if I want to use G02/G03 I have to add a\n\u003e third
parameter to each vertex in the array.\n\u003e * In my approach I didn't
touch the Gcode generation functions, if I want\n\u003e to\n\u003e use
functions.\n\u003e Here are two screenshots, you can test the feature on my
test workspace,\n\u003e link in screenshots:\n\u003e
\u003chttps://cloud.githubusercontent.com/assets/24837471/21677191/18825c1c-
\n\u003e d34a\n\u003e -11e6-96d6-40b96f952ad5.png\u003e\n\u003e\n\u003e
\u003chttps://cloud.githubusercontent.com/assets/24837471/21677197/1f6a0eb2-
\n\u003e d34a\n\u003e -11e6-9f69-f306fcf90079.png\u003e\n\u003e\n\u003e
\u003chttps://github.com/chilipeppr/widget-eagle/issues/13#issuecomment-\n\u
auth/AQgKLXJo319jUJl8IeYGWIlhs\n\u003e Bkt9F3pks5rPMWigaJpZM4LaEPs\u003e
.\n\u003e\n\u003e\n\u003e
{\"api_version\":\"1.0\",\"publisher\":{\"api_key\":\"\n\u003e
05dde50f1d1a384dd78767c55493e4b\n\u003e
b\",\"name\":\"GitHub\"},\"entity\":{\"external_key\":\"github/\n\u003e
chilipeppr/widget-eagle\n\u003e
\",\"title\":\"chilipeppr/widget-eagle\",\"subtitle\":\"GitHub\n\u003e
/\n\u003e 143\n\u003e
418/17495839/a5054eac-5d88-11e6-95fc-7290892c7bb5.png\",\"\n\u003e
avatar_image_url\":\"h\n\u003e
ttps://cloud.githubusercontent.com/assets/143418/15842166/7c72db34-2c0b-\n\u
003e 11e6\n\u003e -9aed-b52498112777.png\",\"action\":{\"name\":\"Open
in\n\u003e 
GitHub\",\"url\":\"https://github.com/chilipeppr/widget-eagle\"}}\n\u003e
,\"updates\":{\"sni\n\u003e
dimensions\n\u003e with curves are now supported, still experimental I need
to do further\n\u003e testing, I implemented the curves as line segments
(currently 8 per each\n\u003e 1/4\n\u003e circle), so no G02/G03 is
generated, I might make the number of segments\n\u003e dynamic depending on
curve size.\\r\\nI decided to take this approach for\n\u003e following
reasons:\\r\\n- I didn't want to mess-up with clipperDimension\n\u003e
G02/G03\n\u003e I have to add a third parameter to each vertex in the
array.\\r\\n- In my\n\u003e approach I didn't touch the Gcode generation
functions, if I want to use\n\u003e G02/G03 I have to rewrite that and
possibly other inflation\n\u003e functions.\\r\\n\\r\\nHere are two
screenshots, you can test the feature on my\n\u003e test workspace, link
in\n\u003e screenshots:\\r\\n![curves_board](https://cloud.\n\u003e
githubusercontent.com/assets/2\n\u003e
4837471/21677191/18825c1c-d34a-11e6-96d6-40b96f952ad5.\n\u003e
png)\\r\\n\\r\\n![curves_g\n\u003e
code](https://cloud.githubusercontent.com/assets/\n\u003e
24837471/21677197/1f6a0eb2-\n\u003e
d34a-11e6-9f69-f306fcf90079.png)\\r\\n\"}],\"action\":{\"name\":\"View\n\u00
3e Issue\",\"url\":\"https://github.com/chilipeppr/widget-eagle/\n\u003e
issues/13#issuecomm\n\u003e ent-270612423\"}}}\n\u003e\n\u003e\n\u003e
\u003chttps://github.com/chilipeppr/widget-eagle/issues/13#issuecomment-2706
-JddSljVqFjTks5rPQjWgaJpZM4LaEPs\u003e\n\u003e
.\n\u003e\n"}],"action":{"name":"View
Issue","url":"https://github.com/chilipeppr/widget-eagle/issues/13#issuecomm
ent-270703008"}}}

ameennihad, 2017-01-10:
When I generate gcode for a curve using G01, what is the best segment length to use? in practice a curve with 1mm long segments looks very smooth on the screen.

momiccioli, 2017-01-10:
I have been using pcb-gcode in eagle.  Here is a sample of a corner it generated.  It uses fairly small increments.  Most likely to ensure a smooth round corner.  The first and last lines are the run outs from the adjoining corners.

G01 X-3.0348 Y0.3072 
G01 X-90.3610 Y0.3072 
G01 X-90.3725 Y0.3049 
G01 X-90.4073 Y0.3072 
G01 X-90.4423 Y0.3072 
G01 X-90.4536 Y0.3102 
G01 X-90.7127 Y0.3272 
G01 X-90.7477 Y0.3272 
G01 X-90.7590 Y0.3302 
G01 X-90.7707 Y0.3310 
G01 X-90.8038 Y0.3422 
G01 X-91.3591 Y0.4910 
G01 X-91.4275 Y0.5094 
G01 X-92.0369 Y0.8612 
G01 X-92.0379 Y0.8623 
G01 X-92.5346 Y1.3589 
G01 X-92.5371 Y1.3633 
G01 X-92.8399 Y1.8878 
G01 X-92.8864 Y1.9683 
G01 X-92.9105 Y2.0582 
G01 X-93.0536 Y2.5920 
G01 X-93.0648 Y2.6251 
G01 X-93.0656 Y2.6368 
G01 X-93.0686 Y2.6481 
G01 X-93.0686 Y2.6830 
G01 X-93.0856 Y2.9422 
G01 X-93.0886 Y2.9535 
G01 X-93.0886 Y2.9884 
G01 X-93.0909 Y3.0233 
G01 X-93.0886 Y3.0348 
G01 X-93.0886 Y63.3005

chilipeppr, 2017-01-10:
I would go with perhaps 24 segments. That could be high, but computers are
pretty amazing these days, disk space is cheap, bandwidth is fast. Instead
of saving space in the text file using less lines, might as well go up in
resolution a bit.

ameennihad, 2017-01-10:
I'm trying to make the number of segments a function of curve length, so that totally round board and small curve at a board corner will have the same resolution but different number of segments per curve.

chilipeppr, 2017-01-10:
Ahh. Well, in Fusion 360 their default is a "Tolerance of 0.1mm" so I'm
guessing they never let the tangent line go further out than 0.1mm. I'm
guessing a little a(2) + b(2) = c(2) could help with a formula like that. I
sometimes change that 0.1mm to 0.01mm if i want a really clean curvy corner
though, but that's rare.

ameennihad, 2017-01-10:
I might expose this value to the UI, so user can adjust  it if required.

momiccioli, 2017-01-10:
> <reply+01080a2dfc4ed7f121831fd8f2eedcb7287698a011b35a8892cf00000001148ccb729

I agree with John add a minor tolerance if for nothing else to keep the
segments from getting out of control.

Date:  Tuesday, January 10, 2017 at 11:18 AM

<https://github.com/chilipeppr/widget-eagle/issues/13#issuecomment-271620581
BCYFjnUks5rQ69ygaJpZM4LaEPs> .

  
ppets":[{"icon":"PERSON","message":"@ameennihad in #13: I might expose this
required."}],"action":{"name":"View
Issue","url":"https://github.com/chilipeppr/widget-eagle/issues/13#issuecomm
ent-271620581"}}}

ameennihad, 2017-01-20:
Complete curves support and curve resolution feature is implemented in V5.3

closed 2017-01-20 by ameennihad