chilipeppr/widget-eagle #14: Pads Diameter

ameennihad, 2017-01-05:
**Bug:**
Pads are sometimes rendered with diameter larger than it should be, see below screenshots of Arduino Uno board, in Eagle the lower pad of USB connector is around 15mil away from the trace on its left, while in Chillipeppr the pad and the trace overlap.
**Cause:**
When pad diameter is set to "auto" Eagle removes the diameter attribute from the pad, in this case the code assumed that diameter is twice the drill size which is incorrect, Eagle uses the following to calculate the diameter:

- If drill size is lower than 0.9mm: diameter = drill + 0.4064
- if drill size larger than 2.1mm: diameter = drill + 1.016
- if drill size >= 0.9mm and <=2.1mm: diameter = drill x 1.5

[image]

[image]

raykholo, 2017-01-05:
In Eagle DRC the section "restring" is what handles this.

[image: Inline image 1]

You should look at some of the values here, from the .brd XML and make a
brd with unique values in the DRC to see what corresponds to what.

> <param name="mdWireWire" value="6mil"/>
>
> <param name="mdWirePad" value="6mil"/>
>
> <param name="mdWireVia" value="6mil"/>
>
> <param name="mdPadPad" value="6mil"/>
>
> <param name="mdPadVia" value="6mil"/>
>
> <param name="mdViaVia" value="6mil"/>
>
> <param name="mdSmdPad" value="0mil"/>
>
> <param name="mdSmdVia" value="0mil"/>
>
> <param name="mdSmdSmd" value="0mil"/>
>
> <param name="mdViaViaSameLayer" value="8mil"/>
>
> <param name="mnLayersViaInSmd" value="2"/>
>
> <param name="mdCopperDimension" value="10mil"/>
>
> <param name="mdDrill" value="10mil"/>
>
> <param name="mdSmdStop" value="0mil"/>
>
> <param name="msWidth" value="5mil"/>
>
> <param name="msDrill" value="12mil"/>
>
> <param name="msMicroVia" value="12mil"/>
>
> <param name="msBlindViaRatio" value="0.5"/>
>
> <param name="rvPadTop" value="0.25"/>
>
> <param name="rvPadInner" value="0.25"/>
>
> <param name="rvPadBottom" value="0.25"/>
>
> <param name="rvViaOuter" value="0.25"/>
>
> <param name="rvViaInner" value="0.25"/>
>
> <param name="rvMicroViaOuter" value="0.25"/>
>
> <param name="rvMicroViaInner" value="0.25"/>
>
> <param name="rlMinPadTop" value="10mil"/>
>
> <param name="rlMaxPadTop" value="20mil"/>
>
> <param name="rlMinPadInner" value="10mil"/>
>
> <param name="rlMaxPadInner" value="20mil"/>
>
> <param name="rlMinPadBottom" value="10mil"/>
>
> <param name="rlMaxPadBottom" value="20mil"/>
>
> <param name="rlMinViaOuter" value="6mil"/>
>
> <param name="rlMaxViaOuter" value="20mil"/>
>
> <param name="rlMinViaInner" value="6mil"/>
>
> <param name="rlMaxViaInner" value="20mil"/>
>
> <param name="rlMinMicroViaOuter" value="4mil"/>
>
> <param name="rlMaxMicroViaOuter" value="20mil"/>
>
> <param name="rlMinMicroViaInner" value="4mil"/>
>
> <param name="rlMaxMicroViaInner" value="20mil"/>
>
>
Why am I telling you this? Because the DRC has a min, max, and percent
value for pads, vias, etc... I tell eagle and the ring around my vias is
always 6 mil if it's on auto.  So in that case your formula wouldn't
apply.  Which is why I think you need to first check these DRC values as a
"sanity check" before the widget decides to "correct the pads" by applying
your formulas.

ameennihad, 2017-01-05:
Thanks ray for pointing this out, apparently the constants in my formula are the defaults, I guess all we need to do is pull those values from the XML file rather than hard code them.
I applied the formula to vias when I did the fix to vias not inflated bug.
For example the constant I'm using for drill > 2.1mm is 1.016 which is 40mils, I guess that would be rlMaxViaOuter+rlMaxViaInner, both are 20mils by default.
Interesting, I'll look into this later and try to find which parameters I should used to replace the constants in the formula.

ameennihad, 2017-01-21:
I changed how vias and pads diameter is calculated, by relying on Restring parameters of Design Rules, I hope @raykholo & @chilipeppr can do some test on [my workspace](http://www.chilipeppr.com/ameen) and let me know if you spot any thing wrong, I couldn't test inner layer because I'm using free version of Eagle.
Also, now square and octagon vias and offset pads are rendering correctly.

**Here is a board that I tested:**

[image]

**Here is how it renders after the update**
[image]

**Here is how it renders on current version of Eagle BRD Widget**
[image]

chilipeppr, 2017-01-21:
Brilliant. Could you send me that Eagle file and I'll post it as a sample
in the file menu. That's a good test case.

closed 2017-02-13 by ameennihad