chilipeppr/widget-eagle #16: PCB Holders Issues

ameennihad, 2017-01-06:
The PCB Holders features has some issues, following is what I noticed:
1. There is a missing Gcode line from one of the corners to the holder on its left, here it seems the cutting tool will plunge deeper than expected in one of the passes, I believe this bug is by design and not related to recent update to board dimension rendering, the result is exactly the same on xpix workspace, I tested Arduino Uno and another rectangular board I designed, the bug is clear on both, see the first two screenshots.
2. There is unnecessary fifth holder on right side of Arduino Uno board.
3. The tabs look very weird on boards with a bit complex shaped dimensions, such boards are rare, but still possible. See third screen shot.

I recommend keeping this feature off by default, until me make it a bit smarter.

**Arduino board on tinyg ws:**
[image]

**My board on xpix ws:**
[image]

**Complex shaped boards**
[image]

chilipeppr, 2017-01-06:
Yes, I agree that this feature is enormously exciting, but it's got bugs. I
also think the tool path should not raise on the Z like it does when it
gets over a tab. Instead it should stay at the same Z and only raise up
when at the last couple of passes.

xpix, 2017-01-07:
Ok, i'll check this and try to fix this.

ameennihad, 2017-01-14:
There is a bug in dimension milling, apparently that no body noticed so far, there is a small gap where the end mil plunges and retracts, it's not obvious because dimension milling starts at a corner which changes to curve after inflation, but now it's very obvious in slots (deflated dimensions), this is the same bug I mentioned above - pint no. 1, I first thought it's related to board holders update, but it's actually caused by Clipper, I noticed that paths; after inflation/deflation; don't start and end at the same X, Y coordinates, it took me a while to figure that out.
I fixed this bug by simply plunging at the last point not the first one.

chilipeppr, 2017-01-14:
Sweet. Good fix. Yeah, I was starting to notice that the start position was
sometimes weird and was thinking perhaps that bug was always there. It's
hard to go from Three.js line pieces to Clipper because they differ in
whether they reiterate the start/end points and you always have to, in
code, make amends for that.

closed 2017-02-13 by ameennihad