chilipeppr/widget-eagle #17: Generate Gcode with a comment on the line above the M6 tool change to help user jump to different sections when in Gcode widget

chilipeppr, 2017-01-06:
[image]

The Gcode widget generates a Tool Changes drop down. It looks at the M6 lines in the Gcode, and then the first comment above the M6 line becomes the label in this dropdown. It would be nicer to have the comment say the section name like "Drilling Vias" or "Milling Traces" as well as the tool diameter.

However, the Gcode widget will also look for Gcode comments of the format:
(T3 D=0.4 CR=0. - ZMIN=-0.5 - FLAT END MILL)

Where the tool number needs to be right after the open parentheses and then it regurgitates that comment in the pulldown menu like this example of Gcode from Fusion 360.

[image]

ameennihad, 2017-01-16:
While I'm fixing the gcode bug I found lately, I decided to make the changes you mentioned here, please let me know if you have any notes. If there are no holes to mill, title of last tool change will only read **Milling Board Dimensions**.

I don't really know what is CR or ZMIN means!

[image]

chilipeppr, 2017-01-16:
Oh, that looks awesome! This is really nice.

ameennihad, 2017-01-20:
Implemented in V5.3

closed 2017-01-20 by ameennihad