chilipeppr/widget-eagle #18: PCB Registration Holes feature

ameennihad, 2017-01-06:
I would like to discuss and find out what is the best approach to implement “PCB Registration Holes” feature! Following is the approach I have in mind:
## Parameters:

- **PCB dimensions**: The width and height of the blank PCB (not dimensions of design). **_Default_**: 100 x 150 (most common size I found on ebay)
- **_Number of holes_**: Either 2 or 4. **_Default_**: 2
- **Holes’ pattern**: Either on corners or centers of edges. **_Default_**: Center
- **Hole diameter**: Common value for all holes. **_Default_**: 3.175mm (1/8”)
- **Tool**: Type and size of bit to be used to drill the holes. **_Default_**: 1/8” drill/end mill.
- **Distance from edges**: The distance from center of hole to edges of the blank PCB, one number to apply to all holes. **_Default_**: 2mm
- **Design Location**: (0, 0) point of the design with respect to blank PCB, this is important when mirroring the bottom layer, we might need to relocate the bottom layer in addition to mirroring. **_Default_**: 6mm from bottom and left corner of PCB (we need to stay away from the pins).

The user should have a way to zero X and Y to a known point with respect to where the registration holes of the machine are located. I think we need to add Offset parameters to set the distance between machine zero and center of registration marks.

## Behavior:

- A blank PCB will be shown in the 3d viewer based on input parameter, and PCB design will be drawn on top of it.
- User hit “Send Gcode” button to send the gcode of drilling the holes to work space. Or possibly to run it directly.
- We can add a button to jog to each individual hole in X and Y only, so user can check if input parameters match machine registration holes.
- Possibility to save and recall parameters should be implemented.

## Question:
Is it better to have this as a feature as part of Eagle BRD widget or make it as a separate widget? I think we should implement it as a separate widget, in that case we need to pass some parameters to Eagle widget to do proper mirror and placement of button layer.

## Sample Images:
[image]

[image]

[image]

[image]

## Process I normally use:
I use the following process whenever there is a need to remove the PCB and put it back in the same place:

- I drill two 1/8” holes on one of the center lines of blank PCB either vertical or horizontal as in the 3rd and 4th images above.
- I built a platform to hold PCBs on my machine, the platform has pre-drilled holes that match the ones I make on the PCBs, and has two touch plates to zero X & Y to (0, 0) point of my design.
- I place the PCB on the platform and use 1/8” pins (broken bits) to fix it in place.

## Image of my platform:
I have 70mmx100mm and 100mmx150mm blank PCBs, so I made the platform based on those sizes.
[image]

chilipeppr, 2017-01-06:
I think this all sounds great. I tend to just tape down the FR4 and pick a
corner and mill, however the only time I've milled double sided I just line
up holes in corners using trial/error by holding down Shift key in 3D
viewer and then jogging to XY positions for those holes by find the G0 Z
negative move into the hole, and using the "Go To" feature of the popup
dialog that shows when you're holding Shift key down.

Your method is much cleaner looking. Since I haven't used it, it's hard to
know what's best, but all of your ideas seem great. Makes me want to make a
PCB holder as well that aligns to my 3040 machine clamping slots. How do
you hold down the FR4 on your holder? Do you just tape it?

I do always try to conserve as much of my FR4 as I can for future milling
runs as I tend to create lots of tiny PCB's rather than ones that use up
the majority of the FR4. It seems in this approach I can't make very
efficient use of my FR4 blanks, but haven't thought it through much.

Not sure this has to be a different widget, but that does feel clean. It
could just be another tab in the current widget. I have found that I need
so much access to the core elements of the Eagle BRD that doing it in the
same widget is easier than trying to expose all of the data via new pubsub
calls. However, doing all of these pull requests can get easier if you just
deploy a new widget and then have a nice standalone code base and can
deploy on your own.

ameennihad, 2017-01-06:
I use painters tape on back side of the board and on the platform, then use double sided tape in between, this makes it much easier to remove the tape, I use the pins for alignment only, I remove them while the job is running. But it's also possible to just use the pins without any tape if the pins fit tightly in both board and platform.

You can efficiently use every part of the FR4 blank, the only waste is the area around the holes, I don't see why this is not possible!

If I decided to implement this feature I'll make it as separate widget at first, I can later port the code to Eagle BRD widget if required.

chilipeppr, 2017-01-06:
Sounds good. Yeah, I was thinking more about this and if we allow
offsetting of your Gcode to move the x,y 0,0 position then i could be
milling out small board on the same larger board and just shifting my x or
y. When i flip, if it's smart enough to know the original shift, it should
know when i flip where the new x,y 0,0 would be and then i'd be golden. I
already want to mill out a pcb holder now to get this new ability.

chilipeppr, 2017-01-07:
I just noticed that you use a G0 move to descend hole drills. It's safer to
use G1 moves that are slower because you are descending into a cut. My G0
move is set at like 3000 mm/min. A better G1 move would be something like
100mm/min.

[image: Inline image 1]

ameennihad, 2017-01-07:
That was a typo, I fixed it and made a pull request.
Gcode for normal drilling with drill bit should be fine, the issue was only with the big holes, right?

chilipeppr, 2017-01-07:
correct. i only saw this on milling drills, not straight drills.

xpix, 2017-01-09:
Hi, some advices from my side:

- i guess it's better to have holes with a maximum distance as possible cuz this make the system more accurate if u flip the board. IMHO two diagonal holes in the corner sounds good.
- to use broken end mills is a good idea, but i ordered some nadle rolls for nadle bearings, here an ebay link http://www.ebay.de/itm/-/302070609836?. The good thing, they are very accurate (~0.002mm in diameter) and very cheap. You can get this guys in different lengths and its able to let them in place on milling
- at my first time i also work with double side tape, but the driller don't like this tape in my experience the top of the driller get some glued tape and you get wrong holes. I use at this time some points with superglue and press this in my vice together with a HPL Base plate, then this will be flat forever :)

Hope this will help :)

ameennihad, 2017-02-14:
Since the widget is now supporting two dimension layers (20 & 46), I'm thinking of implementing this features (PCB Registration Holes) as follows:
- Use layer 46-Milling to cut board dimensions.
- Use layer 20-Dimension to define the dimensions of FR4 board, widget will not generate gcode for this layer but will use it to mirror the board.
- Use vias with special signal names to define the side and location of registration holes, the widget will drill those holes first as first gcode operation, and ignore them when milling the board.

Most of the functionality is already there, I just need to write some code for using vias as registration holes.

Of course I will make everything optional and use can decide whether to use the feature or not.

Here is an example:
[image]

[image]

[image]

chilipeppr, 2017-02-14:
That looks incredible. I think we'll definitely need a video to show folks
how to do it, but this is getting to a huge new level of capabilities to
have this.

ameennihad, 2017-02-14:
I don't have the right tool & experience to make a video, nor now I'll write a tutorial and make it part of wiki page of the widget.

chilipeppr, 2017-02-14:
I can make a video then once it's all published. I'm looking forward to it
because I think it'll be a dream how nice it is.

ameennihad, 2017-02-16:
I completed developing this feature, please check my workspace (http://www.chilipeppr.com/ameen), I canceled the update I made yesterday and started from scratch, I think it's impractical to extract registration holes and size of blank FR4 from BRD file, therefore I put all the settings in the widget.
I just noticed that there is something not right in widget HTML, I'll try to fix that tomorrow and make a
[image]
 pull request.

chilipeppr, 2017-02-16:
Looking sweet. Could I do the merge  this weekend?

ameennihad, 2017-02-17:
This feature is implemented in version 5.4

closed 2017-02-17 by ameennihad