chilipeppr/widget-eagle #20: Milling direction - Conventional vs Climb milling

ameennihad, 2017-01-18:
Have any one though of the direction of milling paths generated by the widget? while testing orientation of paths I noticed that it's random, shouldn't we make all path in the same orientation? does this matter for milling thin layer of copper or cutting fiberglass board? if it matters please share with me your thought of which orientation is best? I'm not an expert in machining but I think climb milling is not recommended when milling width is more than 40% of tool diameter, if this is right we should go for conventional milling since we are milling and cutting in full tool diameter.

From programming perspective, I think we should make all path in the same orientation and give the user the option to select Conventional or Climb milling, and set conventional as default.

chilipeppr, 2017-01-18:
I don't really know best answer on that either. If you are using a chip
breaker end mill to cut out your FR4 then I don't think direction matters.
I have heard that you want to cut with as much of the end mill as possible,
rather than the bottom of it, so that you aren't just wearing out the
bottom edges of the end mill. That could mean doing full depth cuts, but
just slower. Today the user can choose that option by changing the depth of
cut.

For just removing copper I don't think the direction matters much either,
although with respect to a clean finish on the edge of the copper it
probably does. You would want the cutting edge moving into the copper in
the reverse direction the path is taking you. What does happen is with
ground planes you get both directions because of the positive path and then
the negative path. It would be nice to reduce that redundancy, but i found
that a hard algorithm to figure out.

For milling holes, it does seem that your helical stuff works well.

The more features, the better, just not sure bang for the buck.

ameennihad, 2017-01-18:
I've been thinking about reducing the number of paths, so far I managed to avoid tracing board dimensions, I'll keep this enhancement on my TODO list, hopefully I'll find a solution one day.

Back to main subject, correcting path orientation is fairly simple, it took my 5 minutes to update the code, I just need to update the widget.html, I'll add two dropdown lists one for milling traces and the other for cutting board dimensions, each one will have three choices (don't care, conventional milling, climb milling).

xpix, 2017-01-19:
I like this idea, i think that a question what for tools you need for cut
the board or mill signal layers. The best is really to let the user decide
which type of milling he want to use.

*--*
*Frank Herrmann    | Software Entwicklung*
+49 172 600 40 61 | xpixer@gmail.com
Neudorfer Weg.10  | 68753 Waghäusel

ameennihad, 2017-01-19:
Yes Frank, that's exactly what I did, hopefully this and other features and update will be reflected to master widget.

ameennihad, 2017-01-20:
Implemented in V5.3

closed 2017-01-20 by ameennihad