chilipeppr/widget-eagle #22: no top layer

jpadie, 2017-02-11:
**use case:**
place a header on a board and connect (or don't) the pins on the bottom layer.  Then import into CP.  
example board for testing: https://dl.dropboxusercontent.com/u/194358/test.brd

**Result**
there is no ability to select the top layer. 

**Expected result**
the top layer is viewable and the pins are isolated on the top layer.

**Info**
+Ameen Nihad says this is because there are no signals on the top layer.  However the absence of signals does not imply the absence of a conductive surface (two sided Cu board); which obviously requires isolation on both sides.

ameennihad, 2017-02-11:
Through-hole pads should appear in all layers, same for vias, the problem is there is no way to tell which layers are used in a specific board file other than from signal wires, the easy fix is to always populate Top and Bottom layers in Drop Down List (or at least when there are vias or pads), I can't think of a solution for any layers in between.

jpadie, 2017-02-11:
I have never considered issues beyond two-layer boards.  I assume that on mid layers pins get isolated in the same way that a hole would (i.e. no need for an expanded footprint for solder flow). 

in any event, is the answer simply this logic:

1.	assume for any board which has a pin or a via that there is a top and bottom layer.
2.	isolate accordingly
3.	determine what layers are present that have signals.
4.	iterate through the layers and isolate the pin and via on that layer unless the signals are congruent.

ameennihad, 2017-02-11:
Testing if a board has through hole pads/vias would be a bit complex, I just made the Top/Bottom layers available all times, it want harm if user is allowed to look at an empty side of the board.
Please check your single via board on my workspace

closed 2017-02-13 by ameennihad