chilipeppr/widget-eagle #23: inchoate and inconsistent isolation of pins

jpadie, 2017-02-11:
**use case:**
place standard pitch headers on a board and wire them up.  some should be of the same signal and some NC.  
example board here: https://dl.dropboxusercontent.com/u/194358/test.brd

**result:**
depending on the inflation amount, some identical headers are not isolated whereas others are.  this suggests an algorithm inaccuracy or perhaps that the algorithm is taking account of cream.
if one imagines that headers are being used as mechanical stability on a shield you would absolutely not want either the top or the bottom layers of the pins to be connected unless they were the same signal. 
see the following images:

[image]
(0.5 mm inflation - see the conflation of the same signal pins on both sides and the bloc-isolation of all the NC pins)

[image]

(0.2mm inflation - see that the pads are all correctly isolated).

**expected result:**
if a pin is capable of being isolated it should be isolated.  All pins should be isolated properly.  NC pins should be isolated instead of being treated as the same signal.  Importantly taking the example board, if the path inflation is put at 0.5mm (just about feasible for standard headers) then the unconnected pins will be isolated as if having the same signal.  however reduce the inflation to 0.15 and the pins are all properly isolated - which is correct.  

Yes - there is a workaround by giving each pin a signal - but then eagle complains.  and imagine a due or mega board - it doesn't make sense to name every signal when perhaps only 5 are going to be used on the shield.  

**Info/suggestions**
a signal that is 'undefined' should be treated as unique.

ameennihad, 2017-02-11:
Thanks for reporting this issue, I believe the solution is simple, I'll try to fix this in V5.4.

ameennihad, 2017-02-11:
Issue fixed, please test on my workspace

closed 2017-02-13 by ameennihad