chilipeppr/widget-eagle #28: Keep view angle unchanged after refresh

ameennihad, 2017-02-13:
I'm trying to avoid changing the view angle when Refresh or Gcode buttons are pressed, I realized that view angle is set to default when one of the following lines is called:

`chilipeppr.publish('/com-chilipeppr-widget-3dviewer/viewextents' );`
`chilipeppr.publish('/com-chilipeppr-widget-3dviewer/drawextents' );`

What the effect of calling those functions from the 3D viewer?

chilipeppr, 2017-02-13:
That's the equivalent of clicking the eye button in the 3D viewer toolbar.
It could make sense to not have the Refresh or Gcode button trigger those
and let the user manually do that.

ameennihad, 2017-02-17:
I disable "viewextents" call on Refresh, but it's not possible to disable when sending the Gcode, it has to be done from 3D viewer widget.

closed 2017-02-17 by ameennihad