chilipeppr/widget-eagle #30: Backlash issue or path calculation issue?

dgtlmoon, 2017-02-17:
Hi all

Testing the importer using a fairly simple test BRD from EagleCad, I'm using a 60deg V-bit, with the _Inflate Milling Path By_ set to 0.55, the actual path sizing looks great, a couple of things I notice is that the drill positions are way off and theres definitely a problem with the cutting path, ignore the long line that intersects all 3 tests you can see on the big curves that it totally looses its position.

[image]

The final (top) PCB cutting i have flipped X and Y axis in the BRD importer to see what differences I'de see.

Each one has slightly different _"Inflate Milling Path By"_ settings, that seems to work wonderfully.

[image]

And life would not be completed with the Chillipepr test..

[image]

The technical specs for my CNC says "Backlash: approx. 0,08 mm / 0.003 inch(with software adjustable to 0,00 mm)" but obviously I've removed its controller and am using TinyG :)

Attached is my test BRD and GCode output

[mill-test.zip](https://github.com/chilipeppr/widget-eagle/files/783036/mill-test.zip)

Can anyone else reproduce this? Can someone perhaps try to cut that BRD file and reproduce the issue? (happy to buy pizza!)

dgtlmoon, 2017-02-17:
Fixed, sorry guys - was a loose belt on the machine.

[image]

closed 2017-02-17 by dgtlmoon

chilipeppr, 2017-02-17:
Nice work. That looks good although most folks don't mill as deep as you
did. I usually only mill down 0.1mm.

dgtlmoon, 2017-02-17:
Thanks! actually I was nervous about what would happen when it was time to solder - how hard would it be to accidently solder across that gap given theres no solder mask... but turns out all was just fine, that small PCB was just a test piece, thanks for the great project!