chilipeppr/widget-eagle #31: Render button missing

dgtlmoon, 2017-02-22:
"To see mirrored results, either select different board layer or hit **Render** button."

I've flicked through all the tabs and there is no Render button anymore (I remember it was there last week?)

[image]

ameennihad, 2017-02-22:
Render button is still in the same place, but without text just an icon, it's in the sceen shot you attached, at the top right side of the widget.

dgtlmoon, 2017-02-22:
@ameennihad definitely some usability issue, i dont see it, which one is it?

chilipeppr, 2017-02-22:
Perhaps let's add the text back into the button then. I can see why folks
wouldn't know that's what button to hit. Also, you can hit Ctrl+Enter
anywhere in the Eagle widget to trigger refresh, although when I tried it
last it seems to sometimes work and sometimes not.

I do think moving all the way to the render button each time you try a new
setting is frustrating, thus why Ctrl+Enter was added by me a while ago.

dgtlmoon, 2017-02-22:
> 281664085>,

The correct text on the mouseover that has the word "(re)render" would be
fine

ameennihad, 2017-02-23:
Rending the board in 3d viewer is don't pretty fast, the only time consuming task is generating the tool paths for traces, therefore I suggest that we do the following:
- Replace the Render button with "Generate Tool Paths" and relocate it under widget title or under first row of tabs.
- Stop generating tool paths automatically, only do that manually when user click "Generate Tool Paths" button.
- Changing any setting not related to tool paths should refresh the 3d viewer automatically.
- Changing any setting related to tool paths should clear any tool paths generated previously form the 3d viewer, so user knows that he/she need to regenerate tool paths.

chilipeppr, 2017-02-23:
Yeah, that sounds like a much nicer idea. Much more sophisticated and
inline with where the computing power actually takes place.

dgtlmoon, 2017-02-23:
yeah, sounds great. i think move that FR4 option to its own tab - just like how there is the "Solder mask" tab

ameennihad, 2017-02-23:
Blank FR4 Board feature is part of the mirror feature, it should stay in place, Registration Holes feature is causing the confusion and should be place in separate tab.

ameennihad, 2017-02-23:
I see that John has changed the description under Blank FR4 Board, the original text was:

_Following settings will affect how selected board layer is mirrored, if this option is unchecked widget will use board dimensions instead._

Original description makes more sense because the main purpose is layer mirroring.

chilipeppr, 2017-02-23:
Sorry, I didn't mean to change the meaning. I was just trying to clean up
the wording. I suppose even I didn't understand what the meaning was.

ameennihad, 2017-02-23:
I admit that sometimes I don't understand what I wrote a while ago, I hope following screenshots will explain the "Blank FR4 Board" feature better, John please feel free to change the description to more clear and easy to understand text:

1. "Blank FR4 Board" feature enables:
1.1 Top layer selected without mirroring:
[image]
1.2 Bottom layer selected and mirrored
[image]
2. "Blank FR4 Board" feature disabled:
2.1. Top layer selected without mirroring:
[image]
2.2 Bottom layer selected and mirrored
[image]