chilipeppr/widget-eagle #32: Always attempts to drill registration holes, wont create gcode when disabled

dgtlmoon, 2017-02-22:
In the gcode you can see the registration holes comment

[image]

but..

[image]

[image]

If I check `Blank FR4 Board` then uncheck the `Registration Holes` checkbox which is on by default, then uncheck `Blank FR4 Board`, then it gives me an error that I have no gcode to send..

[image]

ameennihad, 2017-02-22:
Of course, because the gcode here is just for drilling the registration holes.
What gcode do you expect to see if you disable registration holes? The gcode for the board can be sent from the small icon on top right side of widget.

dgtlmoon, 2017-02-22:
Ahuh ok, i didnt realise that this tab is ONLY activated when its open, as there are a few tabs that do combined functionality, closing, thanks

closed 2017-02-22 by dgtlmoon

chilipeppr, 2017-02-22:
I do think this layout/structure is getting confusing for folks. I had a
hard time trying to really understand that this Gcode was separate as well.
It's probably best to not put this "Blank FR4 Board" on the 1st tab because
it gives it too much prominence.

dgtlmoon, 2017-02-22:
Yeah, it feels like I'm configuring  some parameter about the board I'm
using rather than a totally optional step