chilipeppr/widget-eagle #33: Multiple isolation passes

erpalma, 2017-02-22:
Hi, I would like to suggest a new feature: calculate multiple passes for wider isolation traces. This is very useful when using a V shaped tool in HV boards where isolation between traces can be quite high and the tool must perform multiple passes. A similar approach is used by flatcam and pcb-gcode.

chilipeppr, 2017-02-23:
One thought is to allow inflation to still be the way it is, but then ask
for the end mill size and calculate multiple paths. The key is to still let
you visualize path overlap. I have found that I'm almost always milling
PCB's where the limiting factor is that I barely have enough clearance even
for one path, thus the need for multiple paths is reduced.

The other thing is, if you do just one isolation path, even if it's thin,
you can achieve a good PCB, but you do need a good solder mask then. I'd
like the DIY community to perhaps find a better way to do solder masks and
then that could be a better approach than milling multiple isolation paths.
Just a thought. I do think the need for multiple paths is still there
though.

In the short term you could just do the 1st inflation, mill, then do a 2nd
inflation and mill again. Not ideal, but at least workable today.

ameennihad, 2017-02-23:
I think we should add an option to generate tool paths of traces based on the clearance specified in the design (Net classes). and even select or suggest tool sizes based on that.
One more thought I have, is to include some sort of tools library in the widget, when a board file is first loaded the widget will look for the best set of tools to used based one clearance in Net classes, diameters of holes ... etc, then the user can select which tools to use for each operation from his/her library.

chilipeppr, 2017-02-23:
That would be amazing. I'm not sure any PCB path creator does that.

erpalma, 2017-02-23:
If I remember correctly pcb-gcode does what ameennihad says. You select a tool dimension and a maximum isolation dimension, then pcb-gcode calculates the number of passes needed based on the clearance specified in the design.

I've milled several PCB with a high voltage (mains) part and that part really need a different clearance wrt a standard logic-level section.

chilipeppr, 2017-02-23:
The solder mask tab does calculate multiple paths and it was quite easy to
do, so this shouldn't be too hard to add. It's really just a matter of
folks having extra time between their day jobs.  The trick on multiple
paths is just running multiple inflation paths and connecting them together
in the final gcode.

erpalma, 2017-02-23:
I'll give an eye to the code to see what can I do!