chilipeppr/widget-eagle #37: Roundness - SMD Pads

ameennihad, 2017-07-23:
SMD pads are always rendered as rectangles because the roundness property is ignored.

[image]

ameennihad, 2017-12-31:
Roundness supported with V5.7

closed 2017-12-31 by ameennihad

chilipeppr, 2017-12-31:
Do you have an example file with round SMDs? I'd like to check it out. I
could use it as well for a screenshot and a Google+ post picture.

ameennihad, 2018-01-01:
I don't have an example of a complete board, I just created one part for testing, I'm working on a board the we can use as an example, while testing the board I noticed a bug/limitation in the original code, SMD pads are always consider being on top layer unless the part is mirrored, in reality it's possible to have an SMD pad on bottom layer. I'll see if I can fix that.

ameennihad, 2018-01-01:
Here is a sample file.
[ESP8266 Ring.zip](https://github.com/chilipeppr/widget-eagle/files/1595906/ESP8266.Ring.zip)

chilipeppr, 2018-01-01:
Awesome. Thanks. This looks great.

[image: Inline image 1]