chilipeppr/widget-eagle #40: Supporting other PCB file formats

ameennihad, 2018-01-02:
What is the possibility of developing a new widget that fires up when a specific file is dropped in the workspace, process the file, convert it to another format and finally pass it to another widget?

With the experimental widget I developed few months ago, the one that handles KiCad PCB files, I had to read the PCB file and create an XML object in memory, then start processing the XML as a PCB, so instead of developing a new widget that does pretty much the same what eagle widget do, I thought it would be easier to create the XML object in Eagle format and pass it to Eagle widget for processing.

Is that doable?
Is it better if I just add the code to eagle widget?

raykholo, 2018-01-02:
We talked about supporting other formats in the past, I suggested an option
might be to parse other formats and create a "dumb eagle file" from them
that can then be passed to the eagle widget.  This way you do not have to
recreate all the ThreeJS and Clipper stuff each time.  I like it, I think
others considered it viable as well.
The main thing is you lose all the "smart" metadata you get from a native
eagle import.  The first consequence that comes to mind (when I was
thinking about importing gerbers) is that your nets would not have names,
so you would have to generate random-unique names for each net to trick the
eagle widget into compliance.  If you are doing native design formats,
however, perhaps more metadata will be available.

You could create a single new widget for multiple formats, or you could
create a unique widget for each additional design suite format you want to
support. Just subscribe it to listen to the appropriate file extension(s)
when ondrop happens.

I specifically think it should be at least one if not multiple separate
widgets. The eagle widget listens for an ondrop file with extension .brd -
we want to keep doing this.
Make a KiCad importer that listens for .kicad_pcb , processes it, and spits
out a converted.brd which the eagle widget will then pick up.

Keep up the awesome work!

ameennihad, 2018-01-02:
How can I pass a file or and object from one widget to another? is there any widget currently doing that?

raykholo, 2018-01-02:
For sure, for example the eagle widget does that for the gCode file it puts out when you hit send to workspace at the end. 

In short, I think you want to do a chilipeppr.publish an ondrop with your output.brd as the payload. This will replicate the user importing a new file, and the eagle widget should automatically grab it. 

Ray

ameennihad, 2018-01-02:
Thanks Ray, I don't know how that slipped my mind!

raykholo, 2018-01-02:
My memory of it is a little hazy, but we dealt with a lot of this writing the STL and 3D Printing widget.  Feel free to ask more stuff. 

Ray

chilipeppr, 2018-01-02:
I would tend to think you should make a new widget. The Eagle widget has
gotten quite large and we've all learned a lot doing it. There's a lot of
code in that widget too that's probably not needed anymore. It would be
nice to start clean, but also steal/borrow as much good code from the Eagle
widget as possible.

However, you could also just make a new widget like you're thinking that
parses KiCad and then creates the Eagle format. What you could do from your
new widget, like Ray suggested, is do a chilipeppr.publish() that is just
the Eagle BRD file you created. Then the Eagle widget would pick up on it.

BTW, to catch drag/drop files in your new widget, just subscribe to the
ondrop event like the Eagle BRD widget does. Then you'll get the publish
event before other widgets. If it's a file your widget wants to handle, it
should return false to the pubsub call and then the pubsub system knows to
not propagate the file any further. Make sure to subscribe with the same
priority level as the Eagle BRD widget.

The 3rd option of just adding to the Eagle widget is possible, but it does
feel a bit dirty to do that.

Here's a 4th option that might be nicer. Make a new widget for KiCad, but
instead of sending in an Eagle file to the Eagle BRD widget, you instead
send in a structured object that matches the structure of the parsed
elements today inside the Eagle BRD widget. Just create a new pubsub inside
Eagle BRD widget that let's you send the structured object into it. You
would just do a subscribe() and make up the subscribe name--maybe something
like "/recvIncomingBoardObject". Then your KiCad widget would publish() to
that name. Just document it at the top of the Eagle BRD widget like the
other pubsub calls so it shows up in the pubsub dialog box for future
reference.

ameennihad, 2018-01-02:
The 4th option sounds nicer indeed, but requires extra work on both widgets, KiCad uses easy to read format for humans but hard to deal with in code, so parsing the file to XML is the best solution I could think of, I already done that in widget-pcb, so I just need to tweak the code to produce an Eagle board file instead of the custom XML format I'm currently using.

xpix, 2018-01-02:
@ameennihad Maybe a good solution, send the content of a dropped kicad file to a server and this will convert the kicad to an eagle file via a REST Service, after this process you can load this result with the origonal eagle widget?

ameennihad, 2018-01-03:
Speaking of widget cleanup, I noticed that we still have links to JSfiddle in the widget, is that okay? Shall I change them to the following?

**Flash:**
http://fiddle.jshell.net/chilipeppr/90698kax/show/light/
http://raw.githubusercontent.com/chilipeppr/element-flash/master/auto-generated-widget.html

**pubsub:**
http://fiddle.jshell.net/chilipeppr/zMbL9/show/light/
http://raw.githubusercontent.com/chilipeppr/widget-pubsubviewer/master/auto-generated-widget.html

**Dragdrop:**
http://fiddle.jshell.net/chilipeppr/Z9F6G/show/light/
http://raw.githubusercontent.com/chilipeppr/elem-dragdrop/master/auto-generated-widget.html

chilipeppr, 2018-01-03:
> master/auto-generated-widget.html
> pubsubviewer/master/auto-generated-widget.html

Please do. I thought I cleaned everything up a while ago, but must have
missed those.