chilipeppr/widget-eagle #42: G3 Commands for full circle

ameennihad, 2018-02-05:
The following fix is causing an issue with TinyG:
https://github.com/chilipeppr/widget-eagle/pull/38

Apparently, G3 commands for full circle are not interpreted correctly by TinyG if there are XY values in the command, here is an example:
G3 X10.64 Y18.93 I0.5000 <- current code after above fix
G3 I0.5000 <- Previous code before fix

I totally forgot how G3/G4 commands work, I need to do some reading to find a solution.

Apparently, not all controllers handle those commands the same way! Is it better to switch back to G1 and breaking the path to multiple segments?

ameennihad, 2018-02-12:
Temporarily fixed in PL #43 will look for better solution later.

chilipeppr, 2018-02-12:
You could break into half circles or 1/3rd circles too. Those seem to be
interpreted in a more consistent way out there. And, of course, G1 moves
are what I always went with because I thought they were brainless, but they
do sure bloat the Gcode.

ameennihad, 2018-02-12:
I thought about breaking into half circles too, most probably that will be the better solutions.