chilipeppr/widget-eagle #44: brd import - text and shapes missing

tarchive, 2019-02-11:
I have been banging my head against the wall this weekend trying to figure this out. I have played around with all the settings in this eagle widget and anything that i thought could be relevant in eagle but nothing has worked. Any ideas? thanks in advanced.

This is test of a single top layer board with rectangle and text in the top layer. 
[image]

What is displayed in chillipeppr after dragging and droping the .brd file.
[image]

raykholo, 2019-02-11:
It’s quite possible that those are not parsed and thus would not be generated. The widget reads the Eagle thing. If you widget indeed does not do it then you can generate Gerbers and make the gCode with FlatCam. 

Ray

chilipeppr, 2019-02-11:
The widget does not support shapes or text. Sorry. Although, if you put
your text into, I think, dimensions, you can mill it out. Maybe try putting
shapes into Dimensions layer too. Maybe it would work.

tarchive, 2019-02-11:
Ray, Having text isn't important, i just added included it so it would be considered too. The rectangles were needed for me to solder pogo pins laying down. Making a rectangle on the top layer made sense at the time but now after sleeping on the idea another day, I thought of a simpler workaround. Just extending the routes past the pads. I can't believe i didn't try this sooner. duh!

[image]

[image]

John, Unfortunately the dimensions layer did not work either.  I did not try every single layer in eagle but i have tried all that i though would be applicable like, pads, vias, milling, ect. 

As you have just seen, i found a work around that will work for me. The bigger question now is, Since this isn't supported, would there any plans to add support for shapes and text?

ameennihad, 2019-02-21:
The widget generate gcode for the following three operations (aside from the solder mask feature):
- Isolate the signals.
- Drill or mill holes.
- Cut board dimensions or cutouts if any.
I don't believe text will ever be supported on the widget, text is meant to be printed as part of the silkscreen printing process, which cannot be done with CNC.
Regarding shapes, only polygons and wires are supported, as far as I remember those are processed in the following cases:
- If polygon/wires connected on a signal layer 1-16.
- Polygons/wires on layers 20 and 46 will be user to define board shape and any cutouts.
- When a polygon intersects with a pad in the same package, the pad will take polygon shape (Good example is the pad used in Arduino LilyPad boards). any polygon inside a package will be ignored if it does not intersect with a pad. Polygon has to be on a signal layer.

As a workaround for text, you can draw it with wires or polygons, it's time consuming and a bit tricky but it should work.